ehong-tl/micropySX126X #3: Question: would you like to have a CircuitPython support as well ?

lyusupov, 2021-04-02:
Hello!

Thank you very much for this MicroPython library !

In my GitHub fork of your repo I've created a port of the library onto Adafruit CircuitPython.
It works fine to me on Nordic Semi nRF52840  board.

This it the patch you may think to look into:
https://github.com/lyusupov/micropySX126X/commit/6e54d9b7cf1147f35cd2f01fc8e26be044b670d4

My question is:
would you like for me to create a Pull Request in order to contribute my changes into your parent repo ?

    Best regards,
    Linar.

ehong-tl, 2021-04-02:
Dear Linar,

Thanks for your contribution for adding CircuitPython support, please create a Pull Request to contribute your changes to parent repo. 

If possible please also note down which hardware you have tested with to README.md.

lyusupov, 2021-04-02:
Thank you!
I've created the PR https://github.com/ehong-tl/micropySX126X/pull/4 to track this activity.

There is no longer need to keep the ticket opened. Closing it right now.

closed 2021-04-02 by lyusupov