ehong-tl/micropySX126X #6: SOS! something error about that!

QiaoTuCodes, 2021-11-09:
hey! i look at your micropySX126X drivers in 2days ago . but i want know how to working that ? if all files push down to mpy-boards . like pyboard_v1.1 . it's not working! return msg say : no memory .... so um ~~ how can i cut driver to mini ver? come on suggest something for me. thanks so much!

ehong-tl, 2021-11-09:
Hi, instead of just copy all the python files into your pyboard, you can try to [compile](https://github.com/micropython/micropython/wiki/Building-Micropython-Binaries) the python files into the firmware then flash it to your pyboard.

QiaoTuCodes, 2021-11-09:
oh! so i just copy the lib all files to micropython firmware freezing folder and compile bin files . flash it to  pyboard  may can used right ?

ehong-tl, 2021-11-09:
Yes, hopefully this will resolve the memory problem, otherwise have to use board that has more memory, i.e. ESP32 based board, etc.

QiaoTuCodes, 2021-11-09:
okey! trying thank you bro!

QiaoTuCodes, 2021-11-10:
hello bro. a new problem say:

Traceback (most recent call last):
  File "main.py", line 4, in <module>
  File "sx1262.py", line 20, in __init__
  File "sx126x.py", line 46, in __init__
TypeError: extra keyword arguments given
MicroPython v1.15 on 2021-11-10: pyBoard Plus with STM32F407VGT6

i checked the 3 files error line codes but nothing. just full copy from your sample. so what happend? how do i doing?

QiaoTuCodes, 2021-11-10:
hello bro. a new problem say:

Traceback (most recent call last):
File "main.py", line 4, in
File "sx1262.py", line 20, in init
File "sx126x.py", line 46, in init
TypeError: extra keyword arguments given
MicroPython v1.15 on 2021-11-10: pyBoard Plus with STM32F407VGT6

i checked the 3 files error line codes but nothing. just full copy from your sample. so what happend? how can do i do that?

QiaoTuCodes, 2021-11-10:
hello bro. a new problem say:

Traceback (most recent call last):
File "main.py", line 4, in
File "sx1262.py", line 20, in init
File "sx126x.py", line 46, in init
TypeError: extra keyword arguments given
MicroPython v1.15 on 2021-11-10: pyBoard Plus with STM32F407VGT6

i checked the 3 files error line codes but nothing. just full copy from your sample. so what happend? how can i do that?

ehong-tl, 2021-11-10:
I think may be in the file sx126x.py, line 46
`self.spi = SPI(0, baudrate=2000000, pins=(clk, mosi, miso))`
`pins` keyword argument might not be compatible with pyboard's micropython.
May be you can try modify the SPI object calling code here and try again.

QiaoTuCodes, 2021-11-10:
oh yeah! great! i'll also think may default useing `from machine import SPI , Pin` maybe change 'from pyb import SPI , Pin' will be fix the compatible. i'll trying... thank you.

QiaoTuCodes, 2021-11-10:
maybe the 2 libs it's not identical

QiaoTuCodes, 2021-11-13:
help sir. a new  problem again:

`Traceback (most recent call last):
  File "main.py", line 4, in <module>
  File "sx1268.py", line 20, in __init__
  File "sx126x.py", line 46, in __init__
TypeError: extra keyword arguments given
MicroPython v1.15 on 2021-11-13; PYBv1.1_CN with STM32F405RG
Type "help()" for more information.
` 
i don't know why? check all codes not change that.  
i just change my Pin ports.

ehong-tl, 2021-11-13:
It is back to the same problem you encountered initially.
I think it's best if you can send me your modified code, let me have a look.

ehong-tl, 2021-11-13:
Paste your modified code here is fine.

QiaoTuCodes, 2021-11-13:
okey zip pack for U. so thank you very much!

QiaoTuCodes, 2021-11-13:
[http://47.106.223.45/modules.zip](url)

ehong-tl, 2021-11-13:
Can you paste your modified sx126x.py code to the comment instead?

QiaoTuCodes, 2021-11-13:
okey!

`from _sx126x import *

from sys import implementation

if implementation.name == 'micropython':
    from machine import SPI, Pin
    from utime import sleep_ms, sleep_us, ticks_ms, ticks_us, ticks_diff

if implementation.name == 'circuitpython':
    import digitalio
    import busio
    from time import sleep, monotonic_ns

    _MS_PER_NS = const(1000000)
    _US_PER_NS = const(1000)
    _MS_PER_S = const(1000)
    _TICKS_MAX = const(536870911)
    _TICKS_PERIOD = const(536870912)
    _TICKS_HALFPERIOD = const(268435456)

    def sleep_ms(ms):
        sleep(ms/1000)

    def sleep_us(us):
        sleep(us/1000000)

    def ticks_ms():
        return (monotonic_ns() // _MS_PER_NS) & _TICKS_MAX

    def ticks_us():
       return (monotonic_ns() // _US_PER_NS) & _TICKS_MAX

    def ticks_diff(end, start):
        diff = (end - start) & _TICKS_MAX
        diff = ((diff + _TICKS_HALFPERIOD) & _TICKS_MAX) - _TICKS_HALFPERIOD
        return diff

class SX126X:

    def __init__(self, cs, irq, rst, gpio, clk='Y6', mosi='Y8', miso='Y7'):
        self._irq = irq
        if implementation.name == 'micropython':
          try:
              self.spi = SPI(2, mode=SPI.MASTER, baudrate=2000000, pins=(clk, mosi, miso))  # Pycom variant uPy
          except:
              self.spi = SPI(2, baudrate=2000000, pins=(clk, mosi, miso))                   # Generic variant uPy
          self.cs = Pin(cs, mode=Pin.OUT)
          self.cs = Pin(cs, mode=Pin.OUT)
          self.irq = Pin(irq, mode=Pin.IN)
          self.rst = Pin(rst, mode=Pin.OUT)
          self.gpio = Pin(gpio, mode=Pin.IN)

        if implementation.name == 'circuitpython':
          self.spi = busio.SPI(clk, MOSI=mosi, MISO=miso)
          while not self.spi.try_lock():
              pass
          self.spi.configure(baudrate=2000000, phase=0, polarity=0, bits=8)
          self.spi.unlock()
          self.cs = digitalio.DigitalInOut(cs)
          self.cs.switch_to_output(value=True)
          self.irq = digitalio.DigitalInOut(irq)
          self.irq.switch_to_input()
          self.rst = digitalio.DigitalInOut(rst)
          self.rst.switch_to_output(value=True)
          self.gpio = digitalio.DigitalInOut(gpio)
          self.gpio.switch_to_input()

        self._bwKhz = 0
[... 1287 lines cut ...]
            sleep_us(1)
            start = ticks_ms()
            if implementation.name == 'micropython':
              while self.gpio.value():
                  yield_()
                  if abs(ticks_diff(start, ticks_ms())) >= timeout:
                      status =  SX126X_STATUS_CMD_TIMEOUT
                      break

            if implementation.name == 'circuitpython':
              while self.gpio.value:
                  yield_()
                  if abs(ticks_diff(start, ticks_ms())) >= timeout:
                      status =  SX126X_STATUS_CMD_TIMEOUT
                      break

        switch = {SX126X_STATUS_CMD_TIMEOUT: ERR_SPI_CMD_TIMEOUT,
                  SX126X_STATUS_CMD_INVALID: ERR_SPI_CMD_INVALID,
                  SX126X_STATUS_CMD_FAILED: ERR_SPI_CMD_FAILED,
                  SX126X_STATUS_SPI_FAILED: ERR_CHIP_NOT_FOUND}
        try:
            return switch[status]
        except:
            return ERR_NONE
`

QiaoTuCodes, 2021-11-13:
i'll not change anything. just change my Pin ports to appay pyboard_v1.1_cn boards.

ehong-tl, 2021-11-13:
Can try to change line 46 to the following code,
`self.spi = SPI(2, baudrate=2000000, sck=Pin(clk), mosi=Pin(mosi), miso=Pin(miso))`

QiaoTuCodes, 2021-11-13:
okey,trying...

QiaoTuCodes, 2021-11-13:
now new problem say:
`ValueError: explicit choice of sck/mosi/miso is not implemented`

ehong-tl, 2021-11-13:
how about this?
`self.spi = SPI(2, baudrate=2000000)`

The pin of your SPI connection should be based on id as stated [here.](https://docs.micropython.org/en/latest/library/pyb.SPI.html#pyb-spi)

In your case id=2 should be, (NSS, SCK, MISO, MOSI) = (Y5, Y6, Y7, Y8) = (PB12, PB13, PB14, PB15)

Sorry, I don't have pyboard in hand, its difficult for me to debug.

QiaoTuCodes, 2021-11-13:
yes , it's 'B12=Y5=NSS \ B13=Y6=SCK \ B14=Y7=MISO \ B15=Y8=MOSI' , so i think `self.spi = SPI(2, baudrate=2000000, sck=Pin(clk), mosi=Pin(mosi), miso=Pin(miso))` match that : Y6=SCK \ Y8=MOSI \ Y7=MISO . it's no problem right?

ehong-tl, 2021-11-13:
sck, mosi and miso arguments are not implemented on micropython port for pyboard, should remove that from SPI object.

Just use `self.spi = SPI(2, baudrate=2000000)`

QiaoTuCodes, 2021-11-13:
okey, try agin.

QiaoTuCodes, 2021-11-13:
guy. new problem here:

`Traceback (most recent call last):
  File "main.py", line 11, in <module>
  File "sx1268.py", line 27, in begin
  File "sx126x.py", line 116, in begin
  File "sx126x.py", line 241, in reset
  File "sx126x.py", line 390, in standby
  File "sx126x.py", line 1271, in SPIwriteCommand
  File "sx126x.py", line 1288, in SPItransfer
TypeError: object with buffer protocol required
`

QiaoTuCodes, 2021-11-13:
i'll changed : 

with sx126x.py code:

line 40 `def __init__(self, cs, irq, rst, gpio):`
line 46 `self.spi = SPI(2, baudrate=2000000)`
line 44 `self.spi = SPI(2, mode=SPI.MASTER, baudrate=2000000)`

with sx1268.py code:

line 19 `def __init__(self, cs, irq, rst, gpio):`
line 20 `super().__init__(cs, irq, rst, gpio)`

ehong-tl, 2021-11-13:
try change the followings,

1. line 1286 to `self.spi.write(bytes([cmd[i]]))`
2. line 1311 to `in_ = self.spi.read(1, write=bytes([dataOut[i]]))`
3. line 1326 to `in_ = self.spi.read(1, write=bytes([SX126X_CMD_NOP]))`
4. line 1340 to `dataIn[i] = self.spi.read(1, write=bytes([SX126X_CMD_NOP]))[0]`

QiaoTuCodes, 2021-11-13:
got it! try again.

QiaoTuCodes, 2021-11-13:
changed say:

Traceback (most recent call last):
  File "main.py", line 11, in <module>
  File "sx1268.py", line 27, in begin
  File "sx126x.py", line 116, in begin
  File "sx126x.py", line 241, in reset
  File "sx126x.py", line 390, in standby
  File "sx126x.py", line 1271, in SPIwriteCommand
  File "sx126x.py", line 1315, in SPItransfer
TypeError: function doesn't take keyword arguments

ehong-tl, 2021-11-13:
try change the following lines,

line 1311 to `in_ = self.spi.read(1, bytes([dataOut[i]]))`
line 1326 to `in_ = self.spi.read(1, bytes([SX126X_CMD_NOP]))`
line 1340 to `dataIn[i] = self.spi.read(1, bytes([SX126X_CMD_NOP]))[0]`

QiaoTuCodes, 2021-11-13:
changed say:

Traceback (most recent call last):
  File "main.py", line 11, in <module>
  File "sx1268.py", line 27, in begin
  File "sx126x.py", line 116, in begin
  File "sx126x.py", line 241, in reset
  File "sx126x.py", line 390, in standby
  File "sx126x.py", line 1271, in SPIwriteCommand
  File "sx126x.py", line 1316, in SPItransfer
TypeError: can't convert bytes to int

ehong-tl, 2021-11-13:
try change the following lines,

line 1311 to in_ = self.spi.read(1, dataOut[i])
line 1326 to in_ = self.spi.read(1, SX126X_CMD_NOP)
line 1340 to dataIn[i] = self.spi.read(1, SX126X_CMD_NOP)[0]

QiaoTuCodes, 2021-11-13:
got it! change right now...

QiaoTuCodes, 2021-11-13:
resay:

Traceback (most recent call last):
  File "main.py", line 11, in <module>
  File "sx1268.py", line 40, in begin
  File "_sx126x.py", line 12, in ASSERT
AssertionError: ERR_INVALID_FREQUENCY

ehong-tl, 2021-11-13:
May I know what frequency you are using?
SX1268 is meant for 410MHz to 810 MHz

QiaoTuCodes, 2021-11-13:
just default frequency . because i want to conn 2 sx1268 modules. just useing RX and TX somthing text will be okey. so i have 2 sx1268 modules here.

ehong-tl, 2021-11-13:
the default example is using sx1262, try to change that and your frequency.

QiaoTuCodes, 2021-11-13:
so like edit right:

sx.begin(freq=`433`, bw=500.0, sf=12, cr=8, syncWord=0x12,
         power=-5, currentLimit=60.0, preambleLength=8,
         implicit=False, implicitLen=0xFF,
         crcOn=True, txIq=False, rxIq=False,
         tcxoVoltage=1.7, useRegulatorLDO=False, blocking=True)

ehong-tl, 2021-11-13:
yes

QiaoTuCodes, 2021-11-13:
i'll changed that:

freq=434.0, bw=125.0, sf=9, cr=7, syncWord=SX126X_SYNC_WORD_PRIVATE,
              power=14, currentLimit=60.0, preambleLength=8, implicit=False, implicitLen=0xFF,
              crcOn=True, txIq=False, rxIq=False, tcxoVoltage=1.6, useRegulatorLDO=False,
              blocking=True

copy from sx1268.py code line 23

but got a new problem say:

Traceback (most recent call last):
  File "main.py", line 11, in <module>
  File "sx1268.py", line 40, in begin
  File "_sx126x.py", line 12, in ASSERT
AssertionError: ERR_SPI_CMD_FAILED

QiaoTuCodes, 2021-11-13:
SX126X_SYNC_WORD_PRIVATE = 0x12

ehong-tl, 2021-11-13:
Can you check your sx1268, does it comes with TCXO? If it does, try to set the correct voltage with `tcxoVoltage,` or disable it by setting to 0 if it is not present.

QiaoTuCodes, 2021-11-13:
oh !!!!! guy! did it! it's work!!! so! cool!

QiaoTuCodes, 2021-11-13:
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE
b'Hello World!'
ERR_NONE

QiaoTuCodes, 2021-11-13:
i'll useing  pyboard_v1.1_cn and pyboard_plus to driver here!

QiaoTuCodes, 2021-11-13:
thank you guy ! you're so nice to me.! did you have chat tools number ? join you friend and show you somthing...:)

ehong-tl, 2021-11-14:
Glad to hear that everything's worked out.

closed 2021-11-14 by ehong-tl