ehong-tl/micropySX126X #7: TypeError: extra keyword arguments given

CajKos, 2022-06-21:
Hi,

I just downloaded the repo and saved it via Thonny Pythin IDE on my Raspberry Pico. However, when I run the code I get the following error message:
`Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "/lib/sx1262.py", line 20, in __init__
  File "/lib/sx126x.py", line 46, in __init__
TypeError: extra keyword arguments given`

I want to communicate between a Raspberry Pi 4 (sender) and Raspberry Pi Pico (receiver) to control a servo motor.
From looking a little into the error, it seems to be connected to CircuitPython, but I am not using CircuitPython but MicroPython for the Pico.

Thanks for any advice!

CajKos, 2022-06-21:
Hi @ehong-tl and thanks for the quick response,

I did as you suggested and the new error message is:
`Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "/lib/sx1262.py", line 20, in __init__
  File "/lib/sx126x.py", line 46, in __init__
ValueError: expecting a Pin
>>> `

is there maybe something wrong in the bigger picture. What I did is that I downloaded the repository and copied the lib and example files to the pico. Do I need to do something else, or have I missed something?

ehong-tl, 2022-06-21:
Hi @CajKos ,

It expects Pin object for sck, miso and mosi argument, you can try modify line 46 to 

`self.spi = SPI(0, baudrate=2000000, sck=Pin(clk), mosi=Pin(mosi), miso=Pin(miso))`

ehong-tl, 2022-06-21:
Hi @CajKos 

Did you specify the correct pin according to your hardware setup when you initialize SX1262 object? i.e.

`sx = SX1262(cs=1, irq=2, rst=3, gpio=4, clk=5, mosi=6, miso=7)`

Please change the number according to your setup.

CajKos, 2022-06-21:
Hi @ehong-tl 

no, but I did it now according to https://www.waveshare.com/pico-lora-sx1262-868m.htm with the pins now being:
`sx = SX1262(cs='P3',irq='P20',rst='P15',gpio='P2')` but it did not change anything.

If I try to run it with the original line `self.spi = SPI(0, baudrate=2000000, pins=(clk, mosi, miso))` I get again the error of "TypeError: extra keyword arguments given"
and if I run it with `self.spi = SPI(0, baudrate=2000000, sck=Pin(clk), mosi=Pin(mosi), miso=Pin(miso))` I get the error of "TypeError: can't convert str to int"

ehong-tl, 2022-06-21:
Hi @CajKos ,

According to Pico pinout, SPI on pin 10, 11 and 12 uses SPI bus 1.

You can try change line 46 again to,

self.spi = SPI(1, baudrate=2000000, sck=Pin(clk), mosi=Pin(mosi), miso=Pin(miso))

CajKos, 2022-06-21:
The result:
"Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
  File "/lib/sx1262.py", line 27, in begin
  File "/lib/sx126x.py", line 115, in begin
  File "/lib/sx126x.py", line 240, in reset
  File "/lib/sx126x.py", line 389, in standby
  File "/lib/sx126x.py", line 1270, in SPIwriteCommand
  File "/lib/sx126x.py", line 1287, in SPItransfer
TypeError: object with buffer protocol required"

ehong-tl, 2022-06-21:
Try change line 1286 to `self.spi.write(bytes(cmd[i]))`

CajKos, 2022-06-21:
"Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
  File "/lib/sx1262.py", line 27, in begin
  File "/lib/sx126x.py", line 115, in begin
  File "/lib/sx126x.py", line 240, in reset
  File "/lib/sx126x.py", line 389, in standby
  File "/lib/sx126x.py", line 1270, in SPIwriteCommand
  File "/lib/sx126x.py", line 1313, in SPItransfer
TypeError: function doesn't take keyword arguments"

ehong-tl, 2022-06-21:
Try change line 1311 to `in_ = self.spi.read(1, dataOut[i])` 
and line 1326 to `in_ = self.spi.read(1, SX126X_CMD_NOP)`
and line 1340 to `dataIn[i] = self.spi.read(1, SX126X_CMD_NOP)[0]`

CajKos, 2022-06-21:
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
  File "/lib/sx1262.py", line 27, in begin
  File "/lib/sx126x.py", line 116, in begin
  File "/lib/_sx126x.py", line 12, in ASSERT
AssertionError: ERR_CHIP_NOT_FOUND

ehong-tl, 2022-06-21:
SPI seems to be working, but it seems like your Pico cannot communicate with the LoRa board, hence no chip found error.

CajKos, 2022-06-21:
Ok, interesting. Any ideas why this could be or how to approach this? I can see that the LoRa board gets power through the Pico

ehong-tl, 2022-06-21:
Have you tried Waveshare's C++ demo code?

CajKos, 2022-06-21:
No because I preferred to be able to do everything in Python.

So if I try that and it works, then the problem could not be the board right?

ehong-tl, 2022-06-21:
> No because I preferred to be able to do everything in Python.
> 
> So if I try that and it works, then the problem could not be the board right?

Yes.

The other way is try to change the SPI baudrate at line 46 to 10000000.

CajKos, 2022-06-21:
Changing the SPI baudrate did not work. I will try the Waveshare C++ demo and then post my results. If you have any other ideas please share.

And thanks for all the help so far!

CajKos, 2022-06-23:
Hi @ehong-tl,
I tried today with another LoRa board (in case the other was damaged) and the new repo. And in the beginning it looked good, because I did not get an error message when I was running the main.py from the RX example. But I did not receive any messages either.
So I connected now the antenna to the board and now I am getting the same error messages as last time
`Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "/lib/sx1262.py", line 27, in begin
  File "/lib/sx126x.py", line 115, in begin
  File "/lib/_sx126x.py", line 12, in ASSERT
AssertionError: ERR_CHIP_NOT_FOUND`
and also
`Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "/lib/sx1262.py", line 27, in begin
  File "/lib/sx126x.py", line 115, in begin
  File "/lib/_sx126x.py", line 12, in ASSERT
AssertionError: ERR_SPI_CMD_TIMEOUT`

ehong-tl, 2022-06-23:
> I tried today with another LoRa board (in case the other was damaged) and the new repo. And in the beginning it looked good, because I did not get an error message when I was running the main.py from the RX example. But I did not receive any messages either.

Did you use the same LoRa settings on both Tx and Rx node?

> So I connected now the antenna to the board and now I am getting the same error messages as last time
> Traceback (most recent call last): File "<stdin>", line 11, in <module> File "/lib/sx1262.py", [...]

It seems to me your Pico is unable to communicate with your LoRa board. Have you tried their C++ demo code?
I've no problem on my side, though I'm not using the same MCU and LoRa board.

CajKos, 2022-06-23:
I am using this LoRa board currently only as Rx because I use SX 1268 470M LoRa Hat [(here)](https://www.waveshare.com/product/sx1268-470m-lora-hat.htm) together with a Raspberry Pi 4 as a sender.

I tried the C++ demo code now. With registering at TTN and setting up an end device. When I then read with putty the output of my Pico the result is:

"Pico LoRaWAN - OTAA - Temperature + LED

Initilizating LoRaWAN ... failed!!!"

The C++ SDK itslef is setup correct as other examples (Blink led) work

CajKos, 2022-06-23:
But how likely is it that 2 new lora boards that arrived just this week are broken? Isn't there a saying that in 90% of computer problems the cause is between chair and keyboard? I must be missing something here, I cannot imagine that both boards are broken.
Does it maybe need an additional power supply for the Lora module?

ehong-tl, 2022-06-23:
Did you plug in the antenna all the time when you power on the LoRa board, because the LoRa board can easily damaged if there's no antenna during power loading.

May be you can try to contact Waveshare technical support for troubleshooting?

CajKos, 2022-06-23:
One one of the boards I plugged in the antenna all the time, on the other I did it later.
Yeah, i can try to contact them. If I get something new, I will post it here.

In any case, thanks for all your help!

CajKos, 2022-07-05:
Hey, for some reason it worked when I tried today and I can send and receive between two Raspberry Pi Picos. I don't know though what was different today.

ehong-tl, 2022-07-05:
Glad everything's worked out for you :)

closed 2022-07-05 by ehong-tl