ehong-tl/micropySX126X #10: Waveshare Pico-Lora-sx1262-868m compatibility

byronczimmer, 2022-08-16:
I've picked up a few of these items:
https://www.waveshare.com/pico-lora-sx1262-868m.htm
And wonder if it would be compatible with this library, since I'd prefer to program them in micropython.

And if so -- what the  sx1262 constructor would be.
I *believe* based on the example in Ping Pong...
sx = SX1262(spi_bus=1, clk=10, mosi=11, miso=12, cs=3, irq=20, rst=15, gpio=2)
that the pinout would be as follows, but I'm unsure of the ?? items.

spi_bus / gp2 ( busy? )
clk / gp10 ( clock )
mosi / gp11
miso / gp12
cs / gp3  ( chip select? )
irq / ??
rst / gp15 ( reset )
gpio / ??

I look forward to your response.

ehong-tl, 2022-08-16:
Sorry, I don't have the Waveshare Pico SX1262 and Pico board to try, but based on previous [closed issues](https://github.com/ehong-tl/micropySX126X/issues/7), this library should be compatible with the Waveshare LoRa board. The default constructor in Ping Pong example should be the correct one.

byronczimmer, 2022-08-16:
Units arrive in the next day or so.
Will let you know.
That other issue definitely helps.  Thank you.

Bra1nsen, 2022-08-24:
iam currently also working with the waveshare board. i would like to connect it with the the things network.

where does AppEUI and AppKey come into play/configuration?

ehong-tl, 2022-08-25:
This library does not support LoRaWAN, it only allows you to handle raw LoRa communication.

byronczimmer, 2022-08-28:
Wanted to share that I got Ping/Pong set up with two of the Waveshare SX1262 Pico Lora Hats on a pair of RPi Picos.
You can add it to your compatibility list. :)

closed 2022-08-30 by ehong-tl