ehong-tl/micropySX126X #11: ERR_SPI_CMD_FAILED

pluzmedia, 2022-11-18:
I am using Ra-01s AI Thinker and using sx1268 but the following error occurred ERR_SPI_CMD_FAILED. MY pins configuration would be 

`sx = SX1268(spi_bus=1, clk=18, mosi=23, miso=19, cs=5, irq=26, rst=16, gpio=4)`

Is my Ra-01s hardware damaged? I have 4 modules and all 4 got the same error. All your help would be much helpful to me. Thanks

ehong-tl, 2022-11-18:
Can try set tcxoVoltage to 0?

pluzmedia, 2022-11-18:
I am getting `ERR_CHIP_NOT_FOUND` even with the default voltage

pluzmedia, 2022-11-18:
The initialization fixed but the `ERR_CHIP_NOT_FOUND` error occurred when sending message frequently say about every 100 milliseconds.

ehong-tl, 2022-11-19:
Could be hardware problem, have you tried the C++ [radiolib](https://github.com/jgromes/RadioLib) library?