ehong-tl/micropySX126X #12: Rapsberry PICOI with Waveshare module issue - ERR_CHIP_NOT_FOUND

fulburncams, 2022-11-23:
Hi,  I am using raspberry pico board with waveshare Pico-LoRa-SX1262-868M module,  and getting the error ...

Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "sx1262.py", line 27, in begin
  File "sx126x.py", line 117, in begin
  File "_sx126x.py", line 12, in ASSERT
AssertionError: ERR_CHIP_NOT_FOUND

... even though it is plugged in LED on etc  i am using files downloaded from your github,  are there any diagnostics i can try

ehong-tl, 2022-11-25:
Hi,

Did you configure to the correct pin?

Have you tried their official example library to test out the device?

fulburncams, 2022-11-25:
Thanks, i updated Pi Os to latest and seems to be working now.  Thanks

closed 2022-11-25 by ehong-tl