ehong-tl/micropySX126X #17: Very low range with two SX1262 modules

balynt2048, 2023-04-20:
Hello,
I have recently purchased 3 pieces of LilyGo T3S3 v1.1 boards with the SX1262. The original arduino firmware on the boards out of the box showed good operation; I set one board to lora send, the other to receive and they could ping with RSSI as good as -33 when right next to each other; this lowered to -80 at about 2 meters.

Then, I flashed the boards to use micropython and managed to get basic messaging working with your driver. But now I see that even when the boards are at 1 cm away, the best RSSI I get is -80, this decreases to -110 at 1 meter, then at about 2 meters, the signal is lost completely.

My guess is that there is something wrong with how I program the SX chip to configure the power amplifier. I read through the begin() and understand that everything should be correct though.  I tried different settings regarding power (should be able to go up to 22mDb), current limit (140mA is the default in the specs), even changing the PA ramp time, no improvement.

Would be grateful for any tips! Thanks!

ehong-tl, 2023-04-25:
Hi @balynt2048 ,

I'm not sure how does the PA is programmed in LilyGo chip, but usually you wouldn't want to input too high power pre-amplified signal into the PA.

For example, E22-400M30S (30 dBm max output), the built-in PA adds around 10 dBm to the input pre-amplified signal. So, in the programming, you shouldn't output signal higher than 20 dBm to the PA.

balynt2048, 2023-05-03:
Hi @ehong-tl ,
Thanks for getting back to me. I also consulted a somewhat official git repo with example c-codes, but they couldn't help much.
Anyways, I managed to make it work! The solution was two-fold:
1. I had to select the correct pins. This was not trivial, because the DIO1 pin of the SX1262 was also connected as the TCXO enable pin which didn't make much sense to me. Furthermore, on my board, there are also RX and TX enable pins.
2. After calling sx.begin(), I had to call sx.setOutputPower(powerlevel) _again_ to actually set the level. Consequent calls to set the power level are all successful. I have no idea why.

After these, everything seems to work at an acceptable power level.

Still, there might be a bug in [setOutputPower](https://github.com/ehong-tl/micropySX126X/blob/d7868f7510bd8a8b16b5a69521af0d7302ba85bd/lib/sx1262.py#L128):
according to the SX1262 documentation, negative power levels should be given as power_level + 256. This makes sense, since the transferred data is an unsigned byte:

> The output power is defined as power in dBm in a range of
> • - 17 (0xEF) to +14 (0x0E) dBm by step of 1 dB if low power PA is selected
> • - 9 (0xF7) to +22 (0x16) dBm by step of 1 dB if high power PA is selected 
 I am able to set these power levels correctly.

ehong-tl, 2023-05-03:
Hi @balynt2048 ,

You are right, my mistake didn't notice the negative value TX power when porting from C code.

I've just updated the master branch with the fix.

Thanks for pointing out.

balynt2048, 2023-05-03:
Thank you for creating this driver!

I think you can add the boards I have tested to the compatibility list, maybe with a link to this issue about the output power setting.

closed 2023-05-08 by ehong-tl