ehong-tl/micropySX126X #22: Sending a message to a specific recipient

KievWolf, 2024-03-06:
There is a need to create a point-to-multipoint network
In which commands would be sent from one device to a specific recipient - for example, “there is one remote control, but you need to turn on different lamps by pressing buttons 1, 2, 3, etc.”
Is it possible to implement this using this library?
I can’t find how to send a message to a specific recipient; moreover, the device address is not written anywhere (unlike the u-lora library).
Thank you

KievWolf, 2024-03-07:
Thank you very much!! All its works!

KievWolf, 2024-03-07:
Maybe you can give me some more advice?
How can I have the recipient's address selected by the operator from the list?
I'm interested in one-way transmission - remote activation of the relay.
There is no need for a return message.
When I accept an address variable from user input, the program is executed only 2 times and ends.

ehong-tl, 2024-03-13:
Each of your end devices has their own hardcoded unique address. 
When the operator sends a message with the targeted address, all of the end devices will do a check to see if the message is intended for them based on the address.
If it is, it will execute your command, else will just ignore the message.
I'm not sure if this is what you are asking for, hope it can help you.

closed 2024-03-13 by KievWolf