ehong-tl/micropySX126X #27: Memory allocation problem with Raspberry pico w

Patxi080, 2024-06-04:
I wanted to use the library in a project but when when the library sx1262 is imported this error happens:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/sx1262.py", line 2, in <module>
MemoryError: memory allocation failed, allocating 176 byte
```
I created a new file to check if it was the library with only:
`import sx1262`
And it keeps happening.
Here is an image:
[image]

ehong-tl, 2024-06-05:
Hi,

This could be due to insufficient RAM in Pico W than in standard Pico.

One thing you can try is to compile the library into .mpy bytecode.

Sorry that I can't give you a complete solution as I have no experience in using Pico / Pico W.

Ruben0595, 2024-06-06:
[sx1262_mpy_lib.zip](https://github.com/user-attachments/files/15605522/sx1262_mpy_lib.zip)

Here are mpy files of the library. just replace the current sx1262/126x .py files with these and should work fine.