ehong-tl/micropySX126X #29: setTCXO needs to be called before calibration is run

Celliwig, 2024-09-27:
In the class SX126X, 'setTCXO' is called after 'config' for both 'begin' and 'beginFSK'. 'config' runs the calibration command. The TCXO needs to be configured before calibration is run. so currently you have:

```
        state = self.config(....)
        ASSERT(state)

        if tcxoVoltage > 0.0:
            state = self.setTCXO(tcxoVoltage)
            ASSERT(state)
```

This needs to be changed to:
```
        if tcxoVoltage > 0.0:
            state = self.setTCXO(tcxoVoltage)
            ASSERT(state)

        state = self.config(....)
        ASSERT(state)
```

This led to a loss of several decibels going by the returned RSSI.

ehong-tl, 2024-09-27:
Hi @Celliwig 

You probably right.

Can you help test it again from the test branch?
https://github.com/ehong-tl/micropySX126X/tree/test

Currently I don't have the hardware in hand for testing.

Celliwig, 2024-09-27:
Sorry, I can't test it I'm afraid. Actually using this library as a basis for a Python implementation on a RPi (I have weird requirements in regards to handling CS lines so used this, rather than some other implementations).

Sorry about that...

closed 2025-12-09 by ehong-tl