ehong-tl/micropySX126X #32: AssertionError: ERR_SPI_CMD_FAILED

Mockedarche, 2025-04-06:
So I've tried a good bit to get this to work. Two different ESP32's, checked continuity with a multimeter, tried a different power source and regardless I always end up getting this error.

Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "sx1262.py", line 27, in begin
  File "sx126x.py", line 131, in begin
  File "_sx126x.py", line 12, in ASSERT
AssertionError: ERR_SPI_CMD_FAILED

Here's the given code
```
from sx1262 import SX1262
import time

sx = SX1262(spi_bus=1, clk=18, mosi=23, miso=19, cs=5, irq=33, rst=14, gpio=26)

#LoRa
sx.begin(freq=923, bw=500.0, sf=12, cr=8, syncWord=0x12,
         power=-5, currentLimit=60.0, preambleLength=8,
         implicit=False, implicitLen=0xFF,
         crcOn=True, txIq=False, rxIq=False,
         tcxoVoltage=1.7, useRegulatorLDO=False, blocking=True)

# FSK
##sx.beginFSK(freq=923, br=48.0, freqDev=50.0, rxBw=156.2, power=-5, currentLimit=60.0,
##            preambleLength=16, dataShaping=0.5, syncWord=[0x2D, 0x01], syncBitsLength=16,
##            addrFilter=SX126X_GFSK_ADDRESS_FILT_OFF, addr=0x00, crcLength=2, crcInitial=0x1D0F, crcPolynomial=0x1021,
##            crcInverted=True, whiteningOn=True, whiteningInitial=0x0100,
##            fixedPacketLength=False, packetLength=0xFF, preambleDetectorLength=SX126X_GFSK_PREAMBLE_DETECT_16,
##            tcxoVoltage=1.6, useRegulatorLDO=False,
##            blocking=True)

while True:
    sx.send(b'Hello World!')
    time.sleep(10)

```

I've quadruple checked my wires and am positive that everything is inline with the code here's a pinout of my esp32 (one of the two)

[image]

my assumption is im doing something simple wrong any guidance would be appreciated.

ehong-tl, 2025-04-06:
Hi @Mockedarche 

Can you try set the tcxoVoltage = 0?

Mockedarche, 2025-04-06:
> tcxoVoltage = 0

Thanks for the quick reply. I have done that and now I get 
Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "sx1262.py", line 27, in begin
  File "sx126x.py", line 137, in begin
  File "_sx126x.py", line 12, in ASSERT
AssertionError: ERR_WRONG_MODEM

ehong-tl, 2025-04-06:
What modem are you using?

closed 2025-04-06 by Mockedarche